Lienol/openwrt-1 #14: 科学上网问题

czxinshui, 2019-12-13:
你好。昨天有幸编译了这个系统，然后对这个系统的【科学上网】的一些问题提出一些建议。
现在这个【科学上网】显示界面相比以前好看了很好，有些小问题希望能接纳。
【基本设置】-【运行状态】中的“百度连接“和”谷歌连接“，能否在选择节点、保存并应用后，直接显示出检测的结果。
【节点类目】-选择自动Ping之后，Ping显示数据的速度很慢，经常不显示。是否可以优化
【节点类目】-选择节点，应用-使用，点击使用之后，弹出的窗口固定在列表第1-2行的中间位置，能否改成悬浮在固定居中的位置或者悬浮在其他位置，这样就不用手动拉上去选择。
【节点类目】-选择节点，应用-使用，点击使用之后，在弹出的窗口，选择TCP_1，超过20秒没反应，看不懂是否有在运行切换节点的进程。

Lienol, 2019-12-13:
第一点还是手动检测比较好，第二点暂时在想办法优化，第三第四点已修复

czxinshui, 2019-12-14:
科学上网【节点名称】-能否添加显示VIP等级，测试了好几个机场。都没显示VIP等级，
科学上网【dns-forwarder 】-使用转发这个模式，无法翻墙，SSR+的插件可以，dns-forwarder 程序是否正常使用

czxinshui, 2019-12-14:
科学上网-【节点列表】--选择节点，应用-使用，点击使用之后，在弹出的窗口，选择TCP_1，超过30秒转到”日志列表“，无运行任何程序，切换节点失败。
测试5次，1次成功，成功切换速度很快，失败过程很慢。

Lienol, 2019-12-14:
你用的什么平台的，我用x64无问题

czxinshui, 2019-12-14:
我用X64的。你多测试

czxinshui, 2019-12-15:
设置每天自动更新订阅地址后，出现这样的情况！
https://s2.ax1x.com/2019/12/15/QWWxhR.png

czxinshui, 2019-12-16:
中午再拉取、更新。编译了一下，现在订阅还是会出现一样的问题！同时请求节点显示机场的名字，还是VIP等级
https://s2.ax1x.com/2019/12/16/Q4X1KS.jpg

Lienol, 2019-12-16:
我这边测试了几个机场订阅，均没复现你这个问题。
你有没有试过先删除所有订阅，再重新订阅？
每家机场订阅格式不同，可能会有不同的个例。

czxinshui, 2019-12-17:
1.
第一次使用19.07科学上网的时候，
填入订阅地址-保存并应用-手动更新，
之后，
再填入一个新机场的订阅地址，把原订阅地址都删除掉--保存并应用-手动更新，
节点列表会把原订阅地址的节点全部删除，换成新订阅地址的节点，
最新编译之后，这个很人性化的功能是否取消掉了？
2.
因为机场订阅地址比较多，请求大佬把节点的机场名称显示出来，还有VIP等级

Lienol, 2019-12-17:
已更新订阅脚本，你可以再尝试下

closed 2019-12-18 by Lienol