Lienol/openwrt-1 #95: Luci status 分頁問題

bob771025, 2020-01-01:
使用最新code(branch: my-19.07-full)
編譯完登入首頁發生問題
[image]

bob771025, 2020-01-01:
另外下make menuconfig 會有以下錯誤
WARNING: Makefile 'package/feeds/luci/luci-proto-modemmanager/Makefile' has a dependency on 'modemmanager', which does not exist
WARNING: Makefile 'package/feeds/packages/meson/Makefile' has a dependency on 'python3-pkg-resources', which does not exist

ldgwfyf, 2020-01-03:
嗯,同楼上一样问题,还有 编译几次后:因体积过大编不出sysupgrade,是不是我操作上有问题?

zcluo, 2020-01-03:
> 使用最新code(branch: my-19.07-full)
> 編譯完登入首頁發生問題
> [image]

同样问题

Lienol, 2020-01-03:
看上去像是缺少依赖。
luci-mod-admin-full 选了吗

zcluo, 2020-01-03:
> 看上去像是缺少依赖。
> luci-mod-admin-full 选了吗

我的是选了

bob771025, 2020-01-03:
預設是有選的

bob771025, 2020-01-03:
找到暫時解決方法
去軟件包那升級一下"luci-mod-status"
就可以解決了
[image]

closed 2020-01-04 by bob771025