Lienol/openwrt-1 #135: 刚刚PULL的代码发现编译错误

chiyu521, 2020-01-09:
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package base-files.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ipv6helper:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package ipv6helper.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for urngd:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package urngd.
 * resolve_conffiles: Existing conffile /home/chiyu/Lienol/build_dir/target-mipsel_24kc_musl/root-ramips/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/chiyu/Lienol/build_dir/target-mipsel_24kc_musl/root-ramips/etc/ppp/options-opkg.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/chiyu/Lienol'
package/Makefile:108: recipe for target '/home/chiyu/Lienol/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/chiyu/Lienol/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/chiyu/Lienol'
/home/chiyu/Lienol/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

Lienol, 2020-01-09:
make clean 重来

chiyu521, 2020-01-09:
OK

closed 2020-01-10 by Lienol