Lienol/openwrt-1 #149: 请教一下这个是什么问题？

But-Always, 2020-01-11:
~/openwrt$ make menuconfig
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..bdbdcfbaf6839567e4c21d8c83f4a1311f769fcc
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..bdbdcfbaf6839567e4c21d8c83f4a1311f769fcc
Collecting package info: doneing...eds/telephony/yatedspire-client-cod-bcg729rss
WARNING: Makefile 'package/feeds/luci/luci-app-unbound/Makefile' has a dependency on 'unbound', which does not exist
WARNING: Makefile 'package/feeds/packages/meson/Makefile' has a dependency on 'python3-pkg-resources', which does not exist
tmp/.config-package.in:121668:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121668:  symbol PACKAGE_vpn-policy-routing depends on PACKAGE_vpnbypass
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121688:  symbol PACKAGE_vpnbypass depends on PACKAGE_vpn-policy-routing
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

bob771025, 2020-01-11:
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a

在make menuconfig前先執行這幾行指令試試

But-Always, 2020-01-11:
> ./scripts/feeds clean
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> 
> 在make menuconfig前先執行這幾行指令試試

这些命令执行过了

bob771025, 2020-01-11:
make -j1 V=ss
這OK嗎?

But-Always, 2020-01-11:
> make -j1 V=ss
> 這OK嗎?

不是编译的时候报错呀。

bob771025, 2020-01-11:
fatal: Invalid revision range ee53a24..HEAD
fatal: Invalid revision range ee53a24..bdbdcfb
fatal: Invalid revision range ee53a24..HEAD
fatal: Invalid revision range ee53a24..bdbdcfb
這個比較有問題
其他的
WARNING: Makefile 'package/feeds/luci/luci-app-unbound/Makefile' has a dependency on 'unbound', which does not exist
WARNING: Makefile 'package/feeds/packages/meson/Makefile' has a dependency on 'python3-pkg-resources', which does not exist

我也有

But-Always, 2020-01-11:
出来的固件能用吗？

bob771025, 2020-01-11:
可以啊
我現在還在build新的阿

But-Always, 2020-01-11:
> 可以啊
> 我現在還在build新的阿

谢谢 那我再试试吧

kang-mk, 2020-01-12:
我也报这个错误，不过编译的固件不影响使用

closed 2020-01-12 by Lienol