Lienol/openwrt-1 #169: 更新之后，编译不成功

czxinshui, 2020-01-14:
make[3]: *** [/home/water/openwrt/build_dir/target-x86_64_musl/trojan-1.14.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory `/home/water/openwrt/feeds/lienol/package/trojan'
time: package/feeds/lienol/trojan/compile#0.43#0.17#4.43
make[2]: *** [package/feeds/lienol/trojan/compile] Error 2
make[2]: Leaving directory `/home/water/openwrt'
make[1]: *** [/home/water/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/water/openwrt'
make: *** [world] 错误 2

kang-mk, 2020-01-14:
2020-01-14T05:44:50.6494139Z See also "/home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt/build_dir/target-x86_64_musl/trojan-1.14.0/CMakeFiles/CMakeOutput.log".
2020-01-14T05:44:50.6508621Z make[3]: *** [/home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt/build_dir/target-x86_64_musl/trojan-1.14.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
2020-01-14T05:44:50.6509131Z Makefile:83: recipe for target '/home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt/build_dir/target-x86_64_musl/trojan-1.14.0/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
2020-01-14T05:44:50.6515475Z make[3]: Leaving directory '/home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt/feeds/lienol/package/trojan'
2020-01-14T05:44:50.6515950Z time: package/feeds/lienol/trojan/compile#0.14#0.04#0.19
2020-01-14T05:44:50.6516499Z package/Makefile:111: recipe for target 'package/feeds/lienol/trojan/compile' failed
2020-01-14T05:44:50.6517624Z make[2]: *** [package/feeds/lienol/trojan/compile] Error 2
2020-01-14T05:44:50.6525498Z make[2]: Leaving directory '/home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt'
2020-01-14T05:44:50.6539486Z package/Makefile:107: recipe for target '/home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
2020-01-14T05:44:50.6540409Z make[1]: *** [/home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2020-01-14T05:44:50.6548256Z make[1]: Leaving directory '/home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt'
2020-01-14T05:44:50.6562623Z /home/runner/work/Beta-OpenWrt-AutoBuild/Beta-OpenWrt-AutoBuild/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-01-14T05:44:50.6566489Z make: *** [world] Error 2
2020-01-14T05:44:50.6569592Z ##[error]Process completed with exit code 2.

同  昨晚之后就一直出这个错误

Lienol, 2020-01-14:
更新过今天最新源码尝试过了吗？

kang-mk, 2020-01-14:
嗯 看到刚刚一个小时前同步了1907源码了   这会刚开始编译  待会看看情况   今天之前尝试了好几次一直都是上面的错误  谢谢大佬辛苦了

czxinshui, 2020-01-14:
我再更新一次试试看

kang-mk, 2020-01-14:
最新的源码已经OK了   编译成功

closed 2020-01-15 by Lienol