Lienol/openwrt-1 #254: 按照教学编译了openwrt的img文件大约35mb，导入虚拟机启动以后找不到network

norikohxb, 2020-01-28:
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
第四步做完编译了以后得到的img导入虚拟机启动以后停留在 
root@(none):/#
然后vi /etc/config/network
找不到文件也就是无法修改ip，请问是什么原因？有人可以提示一下吗？

kang-mk, 2020-01-28:
大约35mb
从这里就开始出问题了，如果你是装在物理机那就没问题
如果你转成虚拟机磁盘，需要手动扩大一下虚拟机磁盘，200M以上就可以了，然后就可以正常启动了
这个问题好像是因为生成的35M仅包含了系统分区大小，没有额外的空闲空间了，导致配置文件无法生成
具体在编译过程中的解决方法我也不清楚了，只知道扩大分区就可以了

norikohxb, 2020-01-28:
请问怎么扩大虚拟机的磁盘，扩大不了，能否在编译这个openwrt 的img的时候就指定大小到200Mb，请问要怎么做？

kang-mk, 2020-01-29:
我是hyper-v虚拟机，可以直接编辑，VM的虚拟机不清楚了

Lienol, 2020-01-29:
删除.config再make menuconfig重新编译

norikohxb, 2020-01-29:
> 删除.config再make menuconfig重新编译

你这个是重新制作菜单，我做出来的都是35Mb，请问在制作img之前要怎么设置才能设置出来大一点的？

Lienol, 2020-01-29:
默认就是160M，你编译出来35M是有问题的，检查自己问题

closed 2020-01-29 by Lienol

norikohxb, 2020-01-29:
你的lean-lede我编译是170mb可以用，我说的是你的openwrt版本我编译以后是35mb你可能误解我的意思了