Lienol/openwrt-1 #270: 最新1月30日防火墙打开之后很久无法进入后台，端口转发也不成功

luobobo896, 2020-01-30:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

Lienol, 2020-02-07:
有没有使用”解锁网易云灰色歌曲“？

Lienol, 2020-02-10:
尝试今天的版本

luobobo896, 2020-02-10:
网易云是正常的，已经尝试今天的版本了，重启正常

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

closed 2020-02-11 by Lienol