Lienol/openwrt-1 #281: 科学上网31号版本

czxinshui, 2020-02-02:
科学上网31号版本 出现无法更新订阅，无法更改节点，删除节点再点保存应用后，全部都在的问题

Lienol, 2020-02-03:
订阅链接发我测试

czxinshui, 2020-02-04:
2号版本也是同样 V2ray  订阅地址出现无法更新订阅，无法更改节点，删除节点再点保存应用后，无效

czxinshui, 2020-02-04:
不是马上发生的。是使用一段时间后才会出现的问题

Lienol, 2020-02-06:
出现问题时，发一下passwall的日志

czxinshui, 2020-02-07:
2020-02-07 16:47:17: 删除所有防火墙规则...
2020-02-07 16:47:18: 清除定时执行命令。
2020-02-07 16:47:18: 关闭相关程序，清理相关文件和缓存完成。
2020-02-07 16:47:19: DNS：不使用，将会直接使用上级DNS！
2020-02-07 16:47:19: dnsmasq：生成配置文件并重启服务。
2020-02-07 16:47:22: TCP节点：🇯🇵 日本 v3 | 加速 | 视频流量 | ①，节点：113.207.102.0:80，监听端口：1041
2020-02-07 16:47:22: 开始加载防火墙规则...
2020-02-07 16:47:22: 默认代理模式：大陆白名单
2020-02-07 16:47:22: IPv4 防火墙TCP转发规则加载完成！
2020-02-07 16:47:22: 设置自动更新规则在每天 4 点。
2020-02-07 16:47:22: 设置节点订阅自动更新规则在每天 3 点。
2020-02-07 16:47:22: 运行完成！

2020-02-07 16:47:29: 删除所有防火墙规则...
2020-02-07 16:47:29: 清除定时执行命令。
2020-02-07 16:47:29: 关闭相关程序，清理相关文件和缓存完成。
2020-02-07 16:47:30: DNS：不使用，将会直接使用上级DNS！
2020-02-07 16:47:30: dnsmasq：生成配置文件并重启服务。
2020-02-07 16:47:33: TCP节点：🇯🇵 日本 v3 | 加速 | 视频流量 | ①，节点：113.207.102.0:80，监听端口：1041
2020-02-07 16:47:33: 开始加载防火墙规则...
2020-02-07 16:47:33: 默认代理模式：大陆白名单
2020-02-07 16:47:34: IPv4 防火墙TCP转发规则加载完成！
2020-02-07 16:47:34: 设置自动更新规则在每天 4 点。
2020-02-07 16:47:34: 设置节点订阅自动更新规则在每天 3 点。
2020-02-07 16:47:34: 运行完成！

修改节点自动切换到原来的节点，还有其他 科学上网 别的操作都不成功

closed 2020-02-09 by czxinshui