Lienol/openwrt-1 #308: firewall 问题

luckyyyyy, 2020-02-07:
从目前的情况来看 似乎是网易云音乐导致的 但是还不确定

主要表现是 锁不释放 firewall 无法进行任何操作 

[image]

eanucc, 2020-02-07:
我似乎也遇到同样问题，passwall在进行"保存应用“时，出现“未保存的配置"提示。
另这两天x86-64编译似乎都通不过。

luckyyyyy, 2020-02-07:
> 我似乎也遇到同样问题，passwall在进行"保存应用“时，出现“未保存的配置"提示。
> 另这两天x86-64编译似乎都通不过。

如果有lsof的话 执行下看看 方便的话可以编译进去

没有的话 在luci上重启下防火墙 然后过一分钟执行下面两条看一下结果 发出来是什么
```shell
ps|grep firewall 
ps|grep flock
```

Lienol, 2020-02-07:
测试了一下，的确像是网易云音乐导致

eanucc, 2020-02-07:
但测试了一下，将网易音乐从luci中卸载，并重启，最后还是解决不了问题。难道要在编译的时候就不编进去？

luckyyyyy, 2020-02-07:
> 但测试了一下，将网易音乐从luci中卸载，并重启，最后还是解决不了问题。难道要在编译的时候就不编进去？

你这个应该是另外的问题

Lienol, 2020-02-07:
改了一下代码暂时可以解决。。。

luckyyyyy, 2020-02-07:
躺下了 明天我也帮看下 顺便可以看看那个ddns的问题

发自我的iPhone

eanucc, 2020-02-07:
编译最新的x86固件，很多页面的"保存并应用”不起作用，显示"..末保存的设置".

luckyyyyy, 2020-02-07:
你这个是另外的问题 我刚编译的 close

发自我的iPhone

closed 2020-02-08 by luckyyyyy