Lienol/openwrt-1 #313: X86-64固件 页面"保存并应用"不起作用

eanucc, 2020-02-07:
刚编译的x86固件，在多个页面点击"保存并应用"，结果不起实际作用。仅在页面右上角显示"末保存的设置"。
[image]
[image]
[image]

eanucc, 2020-02-08:
是不是因为我是从luci界面升级固件的，才会出现这个问题？

Lienol, 2020-02-08:
有可能是serverchan不太兼容
点击“未保存的配置” - 恢复
然后重新配置passwall试试

eanucc, 2020-02-08:
不是的。我单单对passwall进行配置也会出现这种情况。在多个页面都会出现这种情况。
我是x86-64固件直接刷物理机的。难道是硬件问题吗？不应该呀，怎么没有其他人提这个issue。

eanucc, 2020-02-09:
问题找到了，确实是从Luci升级的问题。
如果你非要使用从兼容系统的Luci升级，记得把保留设置的勾去掉，这样才不会出现上述问题。

eanucc, 2020-02-09:
close

closed 2020-02-09 by eanucc