Lienol/openwrt-1 #315: passwall的udp节点打开后，游戏会连不上

qwer1217, 2020-02-08:
我记得之前看到有人说过，2月1日版本还是好的，从2月5号开始就有这个问题，最新的2月8日修复网易云音乐导致的防火墙问题解决后这个问题还在

Lienol, 2020-02-08:
你这个问题与网易云无关
UDP节点类型是什么？用NatTypeTester测试一下

qwer1217, 2020-02-08:
呃，我已经还原回0201的版本了，大致情况就是，ps4检测nat类型失败，正常udp加速后应该是nat2，我的移动网裸连是nat3，游戏里检测不出服务器延迟，也连不上服务器，感觉就像udp整个被封锁了一样。
这种情况在我用过的大部分openwrt固件里都会出现，ssr+，ssr+魔改都这样，精简版固件一般都没这种情况，所以我猜想，是不是0205版本，你添加了什么插件导致udp冲突了？

qwer1217, 2020-02-09:
我猜测是不是udp透明代理端口跟别的插件冲突了，但是据说新版passwall没法改udp透明代理端口了，能再加回来吗？

wzd1016, 2020-02-09:
我刚才也提了一样的问题，又看见这个。这个就是固件的问题，我跟这个人的问题一样，我用的xbox,他用的ps4

qwer1217, 2020-02-09:
> 我刚才也提了一样的问题，又看见这个。这个就是固件的问题，我跟这个人的问题一样，我用的xbox,他用的ps4

很多大神的固件都这个问题，插件一多就这样，只有精简版固件才能正常，我刚才尝试修改了udp的端口也不管用~

wzd1016, 2020-02-09:
安心先用2月1日的

Lienol, 2020-02-10:
尝试今天的版本

qwer1217, 2020-02-10:
> 尝试今天的版本

试了下果然正常了~！修复的真快，敢问大神，可以大概说下是什么原因导致这种情况的吗？

Lienol, 2020-02-10:
> > 尝试今天的版本
> 
> 试了下果然正常了~！修复的真快，敢问大神，可以大概说下是什么原因导致这种情况的吗？

透明网桥

closed 2020-02-10 by Lienol