Lienol/openwrt-1 #325: 新版本游戏模式nat转发有问题

wzd1016, 2020-02-09:
好像是20200209的版本吧，群里面也反映了，我代替大家反映一下。因为我用xbox玩游戏，udp必须用不同的节点，在访问控制里指定Xbox的ip为游戏模式（udp模式），这样xbox走udp的节点nat才显示开放而不是加载中，没用li大神的固件之前，很多恩上杂牌固件就是总是加载中（相当于不可用），2月1日的版本是开放的。但新版本就又不行了。但是我提一句，过去版本有时候也会有不稳定的时候，需要重启或者更换dns模式再换过来才能保证nat是开放的，但是不管怎样，不会出现加载中。因为这个我确定是固件问题，我刷过好多不同固件。

qwer1217, 2020-02-10:
我目前用的0201版本，很稳定，建议使用大陆白名单模式，同时把udp改成所有，配合youtube上it奶爸的smartdns+chinang+passwall教程，效果最好，你可以试试

wzd1016, 2020-02-10:
> 我目前用的0201版本，很稳定，建议使用大陆白名单模式，同时把udp改成所有，配合youtube上it奶爸的smartdns+chinang+passwall教程，效果最好，你可以试试

it奶爸那个视频我也看了，但是我只用一部分，就是国内udp选择三个（自身、114、dnspod)。国外那个我一直没用顺，他视频下面自己也修改自己说的，我保留之前我这么用，把tcp和socks5用同一个节点，同样开chinang，然后用dns2socks+socks5节点，dns地址用默认的几个之一，这么弄国外也很快，因为我试过自定义修改dns地址，国外出不去

qwer1217, 2020-02-10:
> > 我目前用的0201版本，很稳定，建议使用大陆白名单模式，同时把udp改成所有，配合youtube上it奶爸的smartdns+chinang+passwall教程，效果最好，你可以试试
> 
> it奶爸那个视频我也看了，但是我只用一部分，就是国内udp选择三个（自身、114、dnspod)。国外那个我一直没用顺，他视频下面自己也修改自己说的，我保留之前我这么用，把tcp和socks5用同一个节点，同样开chinang，然后用dns2socks+socks5节点，dns地址用默认的几个之一，这么弄国外也很快，因为我试过自定义修改dns地址，国外出不去

passwall跟其他同类插件最大的不同就是游戏模式是没有转发tcp的，只转了udp，所以我之前建议用大陆白名单+udp所有，你国外出不去估计是游戏模式的原因

wzd1016, 2020-02-10:
> passwall跟其他同类插件最大的不同就是游戏模式是没有转发tcp的，只转了udp，所以我之前建议用大陆白名单+udp所有，你国外出不去估计是游戏模式的原因

我感觉是socks5解析国外是比较局限的。您说的udp所有我明白，就是在访问控制里，把游戏机的ip地址设成游戏模式（ps4 xbox）udp tcp转发端口设成所有。关键是我习惯了用socks节点解析国外dns，因为访问youtube快。但系统只给了google dns和open dns选项，我用的open dns。我想设成奶爸那种smartdns形式，结果出不去，不仅如此，假如我设成8.8.8.8,8.8.4.4这种形式，就是比默认多一个，都国外连接失败

qwer1217, 2020-02-10:
> 我感觉是socks5解析国外是比较局限的。您说的udp所有我明白，就是在访问控制里，把游戏机的ip地址设成游戏模式（ps4 xbox）udp tcp转发端口设成所有。关键是我习惯了用socks节点解析国外dns，因为访问youtube快。但系统只给了google dns和open dns选项，我用的open [...]

呃，我明白了，你是只有游戏机设成游戏模式，其他设备都是大陆白名单，我没设游戏模式，直接就是大陆白名单，然后udp所有，这样也能起到游戏加速的作用，同时tcp开网页视频等也不受影响，我辣鸡移动网，ps4开商店都慢的捉急，我完全按照it奶爸的方法设置的smartdns，没有任何问题

Lienol, 2020-02-10:
尝试今天的版本